rust-embedded/svd #18: Better error handling

Emilgardis, 2017-01-08:
See related issue japaric/svd2rust#51
`svd-parser` uses `Option<String>` for `ElementExt::get_child_text`, and an expect is done, avoiding any way of knowing what had gone wrong. I propose for `svd-parser` to use `error-chain` to fix this problem, but also to use a logging crate to make debugging svd errors easier for the user without having to use utilities like `SVDConv.exe` to check for errors in the svd file and to see if the error is by missing implementation in `svd2rust`.

Emilgardis, 2017-01-29:
I have a WIP [branch]( https://github.com/Emilgardis/svd/commits/add_error_checking) that does this. It is not exactly what I would like it to be currently, but it gives me a much better picture of what happened on an error.

Emilgardis, 2017-02-05:
I'm going to wait for japaric/svd2rust#53 and then rebase my branch. I have also implemented #21 on that branch, it should be on a separate pr.

nickray, 2019-08-14:
@Emilgardis: running into this exact issue when trying to `svd2rust` the ARM Musca-B1 SVD. Any chance your branch is still rebaseable?

robamu, 2021-10-29:
I have dug a little bit deeper and I can confirm that the error comes from the `syn`  crate when calling `generate::device::render`. The precise error happens at `convert_svd_register` / `name_to_wrapped_ty`