rust-embedded/svd #33: support <clusters>

japaric, 2017-06-06:
these add namespacing to registers and can be nested

estutzenberger, 2017-07-24:
I have been working on updates to svd for this. Please take a look at the following output and see if matches your expectation for how clusters should be processed.  If the output seems good, I'll push up the changes so that we can review.

A cluster example from nrf52.svd:
`
    <cluster>        
        <name>INFO</name>
        <description>Device info</description>
          <addressOffset>0x100</addressOffset>
          <register>
            <name>PART</name>
            <description>Part code</description>
            <addressOffset>0x000</addressOffset>
            <access>read-only</access>
            <resetValue>0x00052832</resetValue>
            <fields>
              <field>
                <name>PART</name>
                <description>Part code</description>
                <lsb>0</lsb>
                <msb>31</msb>
                <enumeratedValues>
                  <enumeratedValue>
                    <name>N52832</name>
                    <description>nRF52832</description>
                    <value>0x52832</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>Unspecified</name>
                    <description>Unspecified</description>
                    <value>0xFFFFFFFF</value>
                  </enumeratedValue>
                </enumeratedValues>
              </field>
            </fields>
          </register>
          <register>
            <name>VARIANT</name>
            <description>Part Variant, Hardware version and Production configuration</description>
            <addressOffset>0x004</addressOffset>
            <access>read-only</access>
            <resetValue>0x41414142</resetValue>
            <fields>
              <field>
                <name>VARIANT</name>
                <description>Part Variant, Hardware version and Production configuration, encoded as ASCII</description>
                <lsb>0</lsb>
                <msb>31</msb>
                <enumeratedValues>
                  <enumeratedValue>
                    <name>AAAA</name>
                    <description>AAAA</description>
                    <value>0x41414141</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>AAAB</name>
                    <description>AAAB</description>
                    <value>0x41414142</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>AABA</name>
                    <description>AABA</description>
                    <value>0x41414241</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>AABB</name>
                    <description>AABB</description>
                    <value>0x41414242</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>Unspecified</name>
                    <description>Unspecified</description>
                    <value>0xFFFFFFFF</value>
                  </enumeratedValue>
                </enumeratedValues>
              </field>
            </fields>
          </register>
`

Here is the SVD output:
`Cluster(RegisterInfo { name: "INFO", description: "Device info", address_offset: 256, size: None, access: None, reset_value: None, reset_mask: None, fields: None, write_constraint: None, _extensible: () }, RegisterClusterInfo { registers: Some([Single(RegisterInfo { name: "PART", description: "Part code", address_offset: 0, size: None, access: Some(ReadOnly), reset_value: Some(337970), reset_mask: None, fields: Some([Field { name: "PART", description: Some("Part code"), bit_range: BitRange { offset: 0, width: 32 }, access: None, enumerated_values: [EnumeratedValues { name: None, usage: None, derived_from: None, values: [EnumeratedValue { name: "N52832", description: Some("nRF52832"), value: Some(337970), is_default: None, _extensible: () }, EnumeratedValue { name: "Unspecified", description: Some("Unspecified"), value: Some(4294967295), is_default: None, _extensible: () }], _extensible: () }], write_constraint: None, _extensible: () }]), write_constraint: None, _extensible: () }), Single(RegisterInfo { name: "VARIANT", description: "Part Variant, Hardware version and Production configuration", address_offset: 4, size: None, access: Some(ReadOnly), reset_value: Some(1094795586), reset_mask: None, fields: Some([Field { name: "VARIANT", description: Some("Part Variant, Hardware version and Production configuration, encoded as ASCII"), bit_range: BitRange { offset: 0, width: 32 }, access: None, enumerated_values: [EnumeratedValues { name: None, usage: None, derived_from: None, values: [EnumeratedValue { name: "AAAA", description: Some("AAAA"), value: Some(1094795585), is_default: None, _extensible: () }, EnumeratedValue { name: "AAAB", description: Some("AAAB"), value: Some(1094795586), is_default: None, _extensible: () }, EnumeratedValue { name: "AABA", description: Some("AABA"), value: Some(1094795841), is_default: None, _extensible: () }, EnumeratedValue { name: "AABB", description: Some("AABB"), value: Some(1094795842), is_default: None, _extensible: () }, EnumeratedValue { name: "Unspecified", description: Some("Unspecified"), value: Some(4294967295), is_default: None, _extensible: () }], _extensible: () }], write_constraint: None, _extensible: () }]), write_constraint: None, _extensible: () }),`

This seems to contain everything but I am not sure if it can be processed as expected by `svd2rust`.

japaric, 2017-08-01:
Could you open a PR? It's hard for me to evaluate from the output you have posted (`{:#?}` would have been more useful than `{:?}`)

estutzenberger, 2017-08-01:
Created. I pulled `{:?}` from the example in the repo I think. If it's useful, I can change and update the output sample to use `{:#?}`.

djmcgill, 2017-10-04:
This issue can be closed now correct? 
It'll mean work can start on https://github.com/japaric/svd2rust/issues/107

Emilgardis, 2018-03-01:
Closed with #38 and cd5aefa

closed 2018-03-01 by Emilgardis