rust-embedded/svd #45: Support `modifiedWriteValues`

Emilgardis, 2018-03-02:
This element specifies side effects when  writing to the field.

```xml
  <!-- modifiedWriteValuesType specifies the pre-defined tokens for the write side effects -->
  <xs:simpleType name="modifiedWriteValuesType">
    <xs:restriction base="xs:token">
      <xs:enumeration value="oneToClear"/>
      <xs:enumeration value="oneToSet"/>
      <xs:enumeration value="oneToToggle"/>
      <xs:enumeration value="zeroToClear"/>
      <xs:enumeration value="zeroToSet"/>
      <xs:enumeration value="zeroToToggle"/>
      <xs:enumeration value="clear"/>
      <xs:enumeration value="set"/>
      <xs:enumeration value="modify"/>
    </xs:restriction>
  </xs:simpleType>
```

ryochack, 2018-05-01:
Hi, @Emilgardis.
I tried `cargo build` on refacter branch at 489cce86.
It seems that modifiedwritevalues.rs has not been commited.

```sh
$ cargo build
   Compiling svd-parser v0.6.0 (file:///home/ryo/code/workspace/svd/svd)
error[E0583]: file not found for module `modifiedwritevalues`
  --> src/svd/mod.rs:22:9
   |
22 | pub mod modifiedwritevalues;
   |         ^^^^^^^^^^^^^^^^^^^
   |
   = help: name the file either modifiedwritevalues.rs or modifiedwritevalues/mod.rs inside the directory "src/svd"
```

Should I prepare anything to build?

Emilgardis, 2018-05-02:
Oops, I'll fix that immediately, sorry for the delay

Emilgardis, 2018-05-02:
There we go, 465cf35 has the offending file. Thank you for your patience 😄

ryochack, 2018-05-02:
Thanks!
I confirmed that the build error has fixed.

commit 489cce86 (2018-03-30): Implemented modifiedWriteValues, closes #45

also adds enumerating on fields
```text
In peripheral `TIMER0`
In register `CR`
Parsing field #0
In field `EN`
Invalid modifiedWriteValues variant, found oneToFail
```
closed 2018-09-02 by ryankurte