rust-embedded/svd #62: New release of `svd-parser`?

jamesmunns, 2018-11-17:
It has been about a year between the last release (0.6.0), and [quite a few changes have landed](https://github.com/japaric/svd/compare/v0.6.0...HEAD).

These changes currently blocks some progress on `svd2rust`, at least https://github.com/rust-embedded/svd2rust/pull/256, which adds support for `derivedFrom` registers.

Is there anything preventing a release, either as a trivial or minor release?

jamesmunns, 2018-11-17:
CC @wez

jamesmunns, 2018-11-17:
...but please merge #63 first :)

ryankurte, 2018-11-28:
@jamesmunns I think there are a fair few breaking changes in there / this should be a minor release? Sounds good to me otherwise!

ryankurte, 2019-01-05:
Alright, all PR's landed (and we're blocking https://github.com/rust-embedded/svd2rust/pull/256), unless someone raises a reason not to in the next ~24 hours I will make an `0.7.0` release tomorrow.

japaric, 2019-01-10:
I have send crates.io owner invites to @ryankurte and @Emilgardis.