rust-embedded/svd #70: Handle hex numbers

thejpster, 2019-07-16:
The nRF9160 SVD file (https://github.com/NordicSemiconductor/nrfx/blob/master/mdk/nrf9160.svd) has lines like:

```
<dim>0x4</dim>
<dimIncrement>0x4</dimIncrement>
```

I don't know what these mean semantically, but the SVD XSD says they must be "scaledNonNegativeInteger", which includes `0x` and `#` prefixes (https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Utilities/CMSIS-SVD.xsd#L168).

Currently this crate uses `string.parse::<u32>()`, which does `from_str_radix` with a fixed radix of 10 (see https://doc.rust-lang.org/src/core/num/mod.rs.html#4665-4670) . It therefore doesn't parse these hexadecimal numeric values and panics.

```console
$ svd2rust -i nrf9160.svd
thread 'main' panicked at '/home/jpallant/.cargo/registry/src/github.com-1ecc6299db9ec823/svd-parser-0.6.0/src/lib.rs:377 tree.get_child_text("dim").unwrap().parse::<u32>(): ParseIntError { kind: InvalidDigit }', src/libcore/result.rs:999:5
note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
```

As `std` can't parse integers based on their prefix (https://github.com/rust-lang/rfcs/issues/1098) I propose we add an integer parse function which handles all the numeric formats specified in the XSD and use that instead of `string.parse::<u32>()`.

burrbull, 2019-09-05:
Is this still actual?

burrbull, 2021-11-06:
Close as this should be fixed

closed 2021-11-06 by burrbull