rust-embedded/svd #111: Implement common traits

MarcoIeni, 2020-04-16:
from the api guidelines:
https://rust-lang.github.io/api-guidelines/interoperability.html

For me it would be very useful if the `Device` struct had at least PartialEq in order to do tests.

cr1901, 2020-04-20:
@MarcoIeni I ran headfirst into this issue as well. My workaround for now is the following:

```
// Newtype to implement PartialEq on svd::Device without needing
// to fork the crate.
#[derive(Debug)]
struct DeviceWrapper(svd::Device);

impl PartialEq<DeviceWrapper> for DeviceWrapper {
    fn eq(&self, other: &DeviceWrapper) -> bool {
        let dev_self : &svd::Device = &self.0;
        let other_self : &svd::Device = &self.0;

        (dev_self.name == other_self.name) &&
        (dev_self.version == other_self.version) &&
        (dev_self.description == other_self.description) &&
        (dev_self.address_unit_bits == other_self.address_unit_bits) &&
        (dev_self.width == other_self.width) &&
        (dev_self.cpu == other_self.cpu) &&
        (dev_self.peripherals == other_self.peripherals) &&
        (dev_self.default_register_properties == other_self.default_register_properties)
    }
}

...
match svd::parse(&out_svd) {
        Ok(parsed) => { assert_eq!(DeviceWrapper(svd_dev), DeviceWrapper(parsed)); },
        Err(e) => {
            eprintln!("Parsing output file {} during round-trip test failed", out_svd);
            return;
        }
    }
```

_Unfortunately, this is not sufficient for my needs for [msp430_svd](https://github.com/pftbest/msp430_svd) and trying to remove the outdated local copy we have.

It's not possible for me to actually construct peripheral instances at all because the `_extensible: (),` field is private. I _believe_ this would be solved if a `Default` implementation was derived, combined with struct update syntax (`..`)?

Perhaps provide it behind a feature gate?

MarcoIeni, 2020-04-20:
Wow, thank you for the workaround, but I would prefer a proper fix in this repo

> It's not possible for me to actually construct peripheral instances at all 

See #101