rust-embedded/svd #112: error parsing an svd file generated by this library

MarcoIeni, 2020-04-18:
In [this](https://github.com/MarcoIeni/svdtools/blob/riir/tests/svd_parser.rs) test I try to 
1. parse an svd from a file
2. encode that svd to xml
3. parse the xml in step 2

Step 3 panics with the error:
```
thread 'read_and_write' panicked at 'called `Result::unwrap()` on an `Err` value: STM32L4x2 invalid

Caused by:
    invalid digit found in string', tests/svd_parser.rs:17:20
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Since the xml was generated by this library, I think it should be valid.

burrbull, 2020-04-23:
Can you give us more information or place somewhere SVD file that cause an error?

burrbull, 2020-04-23:
Looks like this bug has been already fixed.

Try current `master` or `build` branches .

closed 2020-04-23 by burrbull

burrbull, 2020-04-24:
See #104