rust-embedded/svd #118: Builder Option arguments

MarcoIeni, 2020-05-09:
Why the builders have Option<T> in some parameters? For example:

https://github.com/rust-embedded/svd/blob/6d61af9d5eee3cf4add32ebc3daccd5965efc770/src/svd/peripheral.rs#L100-L103

should be:
```
pub fn version(mut self, value: String) -> Self { 
     self.version = Some(value); 
     self 
 } 
```

I think that there is no point in calling this function with a None, since it is the default value.

burrbull, 2020-08-30:
> I think that there is no point in calling this function with a None, since it is the default value.

That's not so!

If you discover code of parser, you'll see that parsing optional arguments return `res: Result<Optional<T>>`.
Now we only need do `foo(res?)` to pass it to builder.
With your suggestion we will need use
```
if let Some(r) = res? {
    foo(r)
}
```
for each optional argument.
And sometimes cleaning of field can be useful too.

MarcoIeni, 2020-08-30:
Ok, but if you build it from scratch, it is a little bit ugly to do `builder.version(Some(..))`

Anyway I am closing this issue, thanks for the explanation :)

closed 2020-08-30 by MarcoIeni