rust-embedded/svd #130: Next release?

Sympatron, 2020-12-03:
Version 0.9.0 was released over a year ago and (at least) the atsamd-rs community currently uses [a specific commit of a specific branch of some fork](https://github.com/atsamd-rs/atsamd/blob/master/update.sh#L7-L8) of svd2rust, because this version depends on [commit #4d384c5e3f168d4175b276a782a6762ffcd2a8dc of `svd-parser`](https://github.com/gkelly/svd2rust/commit/2bbb60590096bcb67c91f38bedd1f63f98132abe#diff-2e9d962a08321605940b5a657135052fbcef87b5e360662bb527c96d9a615542R45-R47), because it is not released. I think this is unfortunate.

Is there anything blocking a new release of `svd-parser`?

therealprof, 2020-12-03:
Probably just a lack of requests for a new release. We can certainly spin one, probably after reviewing and merging #128.

burrbull, 2020-12-03:
We should decide what to do with big amount of SVD files that are failed to be parsed due to new checks in `svd-parser`.
Disable checks? Fix test samples?
https://github.com/rust-embedded/svd2rust/pull/447

Sympatron, 2020-12-03:
Maybe these additional checks could be optional? So users of svd2rust can opt out or opt in to them.

burrbull, 2021-03-31:
I think we are ready for new release.

closed 2021-04-25 by burrbull