rust-embedded/svd #151: Warnings for strict

burrbull, 2021-05-08:
I'd prefer if this was on runtime and not compiletime. That can be done with either "unstrict" functions, a bool flag to `functions` or with recoverable errors in `Ok` path.

My preference is the latter. 

Something like this

```rust
pub(crate) fn check_name(name: &str, tag: &str) -> Result<Option<E>> {
...
}
```

Or similar

_Originally posted by @Emilgardis in https://github.com/rust-embedded/svd/issues/134#issuecomment-751463248_

burrbull, 2021-05-08:
@Emilgardis It is of course good idea.
But I see one serious problem.
Where should we handle these warnings?
`svd2rust` just use `::parse`, so we need to log them during this function.

Emilgardis, 2021-05-08:
I think it makes sense for svd2rust to make strict checking optional. Thus, if not running in strict mode, just dump to stderr when check would fail but didn't (due to being in unstrict mode). 

this is similar to various deny/allow in rustc lints

Emilgardis, 2021-05-09:
That sounds like a good idea. Personally I like structured logs better, so `tracing` would be my preference.

closed 2021-11-06 by burrbull