rust-embedded/svd #164: Proposal: struct instead of enum for arrays

burrbull, 2021-11-07:
Instead of enum:
```rust
pub enum Register {
    /// A single register.
    Single(RegisterInfo),
    /// An array of registers.
    Array(RegisterInfo, DimElement),
}
```
I propose to use struct:
```rust
struct Register {
    info: RegisterInfo,
    dim: Option<DimElement>,
}
```
where `dim: None` corresponds to `Register::Single`.
The main goal is to simplify serialize/deserialize.

cc @Emilgardis  @therealprof  @adamgreig

Emilgardis, 2021-11-09:
It would also be possible to do this with

```rust
#[cfg_attr(
    feature = "serde",
    derive(serde::Serialize, serde::Deserialize),
    serde(untagged)
)]
pub enum Register {
    /// A single register.
    Single {
        /// Register information
        info: RegisterInfo,
    },
    /// An array of registers.
    Array {
        /// Register information
        info: RegisterInfo,
        /// Array dim
        dim: DimElement,
    },
}
```

I wonder which one is the best

burrbull, 2021-11-10:
> It would also be possible to do this with
>
> ```rust
> #[cfg_attr(
>     feature = "serde",
>     derive(serde::Serialize, serde::Deserialize),
>     serde(untagged)
> )]
> pub enum Register {
>     /// A single [...]

Unfortunately the result is different from I expect.
[image]

burrbull, 2021-11-10:
I can take identical result with:
```rust
#[derive(Clone, Debug, PartialEq)]
#[cfg_attr(
    feature = "serde",
    derive(serde::Serialize, serde::Deserialize),
    serde(untagged)
)]
pub enum Register {
    /// A single register.
    Single(RegisterInfo),
    /// An array of registers.
    Array(RegisterArray),
}

#[derive(Clone, Debug, PartialEq)]
#[cfg_attr(feature = "serde", derive(serde::Serialize, serde::Deserialize))]
/// ...
pub struct RegisterArray {
    #[cfg_attr(feature = "serde", serde(flatten))]
    /// ...
    pub dim: DimElement,
    #[cfg_attr(feature = "serde", serde(flatten))]
    /// ...
    pub info: RegisterInfo,
}
```
But does this make sense?

burrbull, 2021-11-10:
Let's just merge #173 instead.

Emilgardis, 2021-11-10:
The reason for the difference is the ordering, placing dim before info in the enum struct would solve that.

Does

Emilgardis, 2021-11-10:
Ah, my bad, I see the issue now. Too bad, I dont think there is a good way to solve this with current serde features

closed 2021-11-10 by Emilgardis