rust-embedded/svd #226: Unknown access variant 'read-write ' found

kossnikita, 2023-09-06:
Unnecessary space in access variant `read-write ` results to error

```sh
[2023-08-28T07:35:10Z ERROR svdtools::cli] by svdtools (0.3.0)

    Caused by:
        0: In device `AT32F415xx_v2`
        1: In peripheral `ERTC`
        2: In register `BPR20DT`
        3: Parsing unknown access at 7318:11
        4: unknown access variant 'read-write ' found
```

Also does it make sense to handle such issue for all tags by default? I have to use patches like this:

```yaml
SPI1:
    CTRL2:
      _modify:
        "MDIV3EN ":
          name: MDIV3EN
```

burrbull, 2023-09-17:
@kossnikita could you try to reinstall `svdtools` from source with:
```
cargo update
cargo install --path .
```
and test it?

kossnikita, 2023-09-17:
@burrbull looks fine to me

closed 2023-09-17 by burrbull