rust-embedded/svd #237: M302: Size of Register must be 8, 16 or 32 Bits

kossnikita, 2023-10-25:
From [CMSIS SVDConv docs](https://arm-software.github.io/CMSIS_5/SVD/html/svd_SVDConv_pg.html#:~:text=Size%20of%20Register%20%27NAME%27%3A%27NUM%27%20must%20be%208%2C%2016%20or%2032%20Bits)

<html>
<body>

Message Number | Type | Message Text 
-- | -- | --
M302 | ERROR | Size of Register 'NAME':'NUM' must be 8, 16 or 32 Bits

</body>
</html>

This limitation is not obvious to me. For example, it is easier to create a 12-bit register for a 12-bit ADC than a 32-bit register with only one field.
svd2rust handles this by creating an unsafe method for working with such registers. But CMSIS prohibits such svd files. Should utilities handle this error?

burrbull, 2023-10-25:
Yes, make sense in `svd`.

burrbull, 2023-11-30:
related to #247