rust-embedded/svd #242: `isDefault` does not work as expected and failes to parse.

liamkinne, 2023-11-15:
In my SVD file an enumerated value of a field is labeled with `<isDefault>true</isDefault>`, which seems okay as per the [spec](https://arm-software.github.io/CMSIS_5/SVD/html/elem_registers.html#elem_enumeratedValue). When this is parsed I get an error like:

```
[ERROR svd2rust] Error parsing SVD XML file
    
    Caused by:
        0: In device `PAC52XX`
        1: In peripheral `SYSCLK`
        2: In register `SCCTL`
        3: In field `CLKIN`
        4: Parsing enumeratedValue `ROSC` at 57:17
        5: `EnumeratedValue error: EnumeratedValue must contain one of `value` (passed Some(0)) or `is_default` (passed Some(true)) tags
```

Changing the value from `true` to `1` gives the error:

```
[ERROR svd2rust] Error parsing SVD XML file
    
    Caused by:
        0: In device `PAC52XX`
        1: In peripheral `SYSCLK`
        2: In register `SCCTL`
        3: In field `CLKIN`
        4: Parsing enumeratedValue `ROSC` at 57:17
        5: `EnumeratedValue error: EnumeratedValue must contain one of `value` (passed Some(0)) or `is_default` (passed Some(true)) tags
```

However, changing the value to `1.0` does work. Seems to be an issue where something expects an integer.

Here is the SVD file in question: [pac22140.svd.txt](https://github.com/rust-embedded/svdtools/files/13366186/pac22140.svd.txt)

burrbull, 2023-11-15:
moved to `svd` repo

burrbull, 2023-11-15:
`enumeratedValue` should contain `<value>` OR `<isDefault>`. Not the both.

Emilgardis, 2023-11-15:
as @burrbull points out, spec says

```
<xs:choice>
  <xs:element name="value" type="enumeratedValueDataType"/>
  <xs:element name="isDefault" type="xs:boolean"/>
</xs:choice>
```

liamkinne, 2023-11-15:
I see. Thanks @burrbull and @Emilgardis.

Doesn't surprise me with the myriad of other problems with this SVD.

burrbull, 2023-11-15:
@Emilgardis svd2rust still not support `isDefault`: https://github.com/rust-embedded/svd2rust/issues/172

burrbull, 2023-11-22:
@Emilgardis after some thinking `choice` restriction looks strange as it would be helpful if `default` EnumeratedValue has `value` too to use it in field writers in svd2rust.
Should we move this check in `strict` mode?

Emilgardis, 2023-11-22:
> isDefault | Defines the name and description for all other values that are not listed explicitly.

Looking over the spec, I'm not even sure what `<isDefault>` is supposed to be used for, I don't see the value for it, not even in a debugger.

svdconv doesn't handle isDefault, with or without a value it just removes the enums that don't specify a name

burrbull, 2023-11-22:
Added example there: https://github.com/rust-embedded/svd2rust/pull/767#issue-2004728412

Emilgardis, 2023-11-22:
I see, so it's if no matched enum value, use this. What does `<isDefault>` with a value mean then? If not matched or is `<value>`

burrbull, 2023-11-22:
> What does `<isDefault>` with a value mean then?

It's could be used to manually specify what exact value to use for writer. Technically it is not needed if it can be found automatically. See `minimal_evs_hole` in PR.

PR #245 (2023-11-22): move EnumeratedValue check to strict mode
Closes #242
closed 2023-11-22 by Emilgardis