rust-embedded/svd #249: svd-parser v0.14.3 fails to compile with svd-rs version v0.14.2 and v0.14.3

Tiwalun, 2023-11-26:
Hi,

I'm seeing a problem where v0.14.3 of the svd-parser fails to compile:

    error[E0425]: cannot find function `expand` in module `cluster`
       --> /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/svd-parser-0.14.3/src/expand.rs:322:31
        |
    322 |             for c in cluster::expand(&info, &dim) {
        |                               ^^^^^^ not found in `cluster`
        |
    help: consider importing this function through its public re-export
        |
    3   + use crate::expand;
        |
    help: if you import `expand`, refer to it directly
        |
    322 -             for c in cluster::expand(&info, &dim) {
    322 +             for c in expand(&info, &dim) {
        |

(from https://github.com/probe-rs/probe-rs/pull/1897)

These are the dependencies:
```
 svd-parser v0.14.3
 585   │ │   ├── anyhow v1.0.75
 586   │ │   ├── roxmltree v0.18.0
 587   │ │   │   └── xmlparser v0.13.5
 588   │ │   ├── svd-rs v0.14.2
 589   │ │   │   ├── once_cell v1.18.0
 590   │ │   │   ├── regex v1.10.1 (*)
 591   │ │   │   └── thiserror v1.0.49 (*)
 592   │ │   └── thiserror v1.0.49 (*)
```

After updating `svd-rs` to version v0.14.4 it compiles again.

This seems to indicate that `svd-parser` needs to explicitly depend on the newer version of `svd-rs`.

burrbull, 2023-11-26:
Look like I've updated `dev-dependencies` instead of `dependencies`:
https://github.com/rust-embedded/svd/blob/c1bc8dbcba9953e961f1b4f8b245e50e0b840610/svd-parser/Cargo.toml#L21-L30

burrbull, 2023-11-26:
make PR, please. I'll release `0.14.4` and yank `0.14.3`

PR #250 (2023-11-26): fix: Update svd-rs dep to 0.14.4
Fixes #249
closed 2023-11-26 by burrbull