rust-embedded/svd #254: Incorrect "out of range" values displayed with <enumeratedValue>

DualTachyon, 2023-12-30:
I'm using a project that uses this SVD crate and I found that the error message displayed with an out-of-range value is misleading.

I'm currently using this tool from this repository: https://github.com/bjoernQ/svd2html
I ran this tool against an SVD I'm generating from a bunch of .def files. This tool found some issues with my SVD, but displays a slightly incorrect message:

```
    3: `EnumeratedValue error: Value 4 out of range 0..4', src/main.rs:39:39
```

As you can see, "Value 4" is inside "range 0..4", so this shouldn't be an error. The error that should have been printed is "range 0..3".
If you inspect the SVD File around the line it complains about, you will find a "<bitWidth>2</bitWidth>" which translates to the correct range 0..3.

You can find my SVD at this repository: https://github.com/DualTachyon/rk3588-svd

Please checkout a specific commit that contains the unfixed SVD:
```
$ git checkout 044f00b1f244fc4502c228e3b29c7bd930e8b0cd
```

After building the svd2html tool, you can reproduce the issue in the following way:
```
$ ./target/debug/svd2html --input $(folder_to_my_svd)/rk3588.svd
```

I don't know rust much, so I was unable to figure out how to fix this myself and send a PR.

Let me know if you need further clarifications.

burrbull, 2023-12-30:
Correct error message:
```
❯ svdtools html ./html rk3588.svd 
Processing rk3588.svd
thread 'main' panicked at src/html/html_cli.rs:501:41:
called `Result::unwrap()` on an `Err` value: In file rk3588.svd

Caused by:
    0: In device `RockChip_RK3588`
    1: In peripheral `FSPI`
    2: In register `CTRL0`
    3: Parsing field `IDLE_CYCLE` at 104982:7
    4: `EnumeratedValue error: Value 4 out of range 0..4
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

burrbull, 2023-12-30:
I would propose you to try `svdtools` instead. At least until you fix all SVD bugs. There a lot of them.

DualTachyon, 2023-12-30:
Maybe I misunderstood your message, but how is "Value 4 is outside 0..4" in your post correct? The range of values is 0..3, not 0..4.

burrbull, 2023-12-30:
See line `104982` in SVD. There `bitWidth = 2` with EnumeratedValue s from 0 to 15.

burrbull, 2023-12-30:
> The range of values is 0..3, not 0..4.

Oh. I see. It just we use open range (a..b) and close range (a..=b) in Rust.

burrbull, 2023-12-30:
Error message is here:
https://github.com/rust-embedded/svd/blob/50ee4f7765110321355f3bf155e9173c97084f35/svd-rs/src/enumeratedvalue.rs#L43

DualTachyon, 2023-12-30:
Ah, thanks for clarifying the open/close range thing. I wasn't aware of that.

I was aware of my bitwidth bug as stated in my first post. I was just pointing out that the open range is a bit misleading, not that the parser didn't find a bug on my side. I fixed the range issue before filing this issue. As for the other bugs (_ as name), I haven't fixed those yet due to a lack of names to use.

The real crime is that ARM Development Studio (= the company that owns the svd format), is happy with the bad bit width, so if I hadn't used the other html tool I wouldn't have been any wiser.

burrbull, 2023-12-30:
You could use `svdtools patch` after DEF to SVD convert to automatize bug patching.

burrbull, 2023-12-31:
Fixed by #255

closed 2023-12-31 by burrbull